OLIMEX/NB-IoT #1: How to send AT command to BC66?

shmrymbd, 2019-11-04:
Hi,
Could you share documentation how to send AT command directly to nbiot module?

DanKoloff, 2019-11-06:
You can do that via the USB. You need just USB micro cable, PC and serial terminal.

Connect USB cable between your computer and board's micro USB connector. Check how the NB-IoT-DevKit listed in your computer (for example, if using Windows, open "Windows Device Manager" and find device "USB-SERIAL CH340"). If it is not properly listed or if it is unrecognized make sure to install the CH340T drivers, but this is usually not required.

If everything is alright, remember the name and number of port it got associated with. Install free serial terminal software, I use puTTy and start serial connection on the port that the board is located at. This is how it looks here:

[image]

Start sending AT commands. Search online for the document that contains the list of AT command for BC66.

closed 2019-11-06 by DanKoloff

shmrymbd, 2019-11-06:
Hi,
1. Thank you for your answer. i have done that and can upload skecth. my issue is to talk direct using AT command right via serial terminal? Currently the code is empty and no serial output. do i need to load some sketch via platform.io to the device?

2. I connect 3.7 rechargable battery after doing programming using usb and the code work but
standalone battery the device not bootup and connect to nbiot network. no led on even i write void setup to turn on led. pls advice because manual is not complete.

3. I have demo tommorow to show running on battery alone. potential 2500u bought from you.

Thank you.

BR
Shamry

cskocsis, 2019-11-06:
Netlight only works with + 5V_EXT or USB power.
[image]

shmrymbd, 2019-11-07:
Hi
I used battery only. yes i agree there will be no net light.

I have the problem the code not running when using battery
 i set void setup… led pin high..

should i remove serial print so code not running if no usb serial? this is my code

#include <Arduino.h>
#include <nbClient.h>
#include <ArduinoJson.h>
#include "RTClib.h"
#include <PubSubClient.h> //

nbClient tcp;
//157.230.38.63  cloud.loranet.my
//219.93.105.37  iot.pernec.com.my
IPAddress server(157, 230, 38, 63); // 157.230.38.63  cloud.loranet.my

int sensor = 9;              // the pin that the sensor is atteched to
int state = LOW;             // by default, no motion detected
int val = 0;                 // variable to store the sensor status (value)

int COUNT = 0;
#define deviceID 001010

void callback(char *topic, byte *payload, unsigned int length)
{
  Serial.printf("[RECEIVE] (%s) %.*s\n", topic, length, payload);
}

PubSubClient client(server, 1883, callback, tcp);

void mqtt_setup()
{
  Serial.println("Start Connection MQTT");
  if (client.connect("xxxxxx", "lorafora", "foralora"))
  {
    // client.publish("pernec/celcom/pos/telemetry/POSBOX-004/", "connected");
    client.publish("pernec/celcom/pos/attribute/POSBOX-004/", "online");
    Serial.println("publish...");
    client.subscribe("inTopic");
    Serial.println("subscribe...");
  }
}

void setup()
{
  String imei;
  Dev.noSleep();
  Serial.begin(115200);
  Serial.printf("POSBOX %s\n", Dev.getVersion());
  pinMode(LED, OUTPUT);
  pinMode(12, OUTPUT);    // initialize sensor as an input
  pinMode(sensor,INPUT_PULLDOWN);    // initialize sensor as an input

  Dev.getImei(imei);
  Serial.printf("IMEI %s\n", imei.c_str());
  Dev.waitSimReady();
  Serial.println("Sim Ready");

  Dev.setBand(8);
  Dev.waitCereg();
  delay(200); // must
  Serial.println("Net Ready");
  mqtt_setup();
  digitalWrite(12, HIGH);
}

int check()
{
  int t = millis();
  int getReceiveLevel = Dev.getReceiveLevel();
  Serial.printf("Rx level     %d dbm\n", getReceiveLevel);
  Serial.printf("Rx quality   %d\n", Dev.getQuality());
  Serial.printf("Rx access    %d\n", Dev.getAccess());
  Serial.printf("Cell cid     %d\n", Dev.getCid());
  Serial.printf("Cell tac     %d\n", Dev.getTac());
  char mlts[322];
  Dev.getMlts(mlts, 322);
  Serial.printf("Cell mlts    %s\n", mlts);
  return(getReceiveLevel);
  Serial.printf("Elapsed: %d mili seconds\n", millis() - t);
}

void execute()
{
  int temp = check();
  COUNT = COUNT + 1;
  // String mystring = "{\"count\":\""+ String(COUNT) + "\",\"rssi\":\"" + String(temp) + "\",\"ts\":\"" + String(1573023292)+"\"}";
  String mystring = "{\"count\":\""+ String(COUNT) + "\",\"rssi\":\"" + String(temp) + "\"}";
  Serial.println(mystring);

  char buf[200]; // Put whatever length you need here
  String(mystring).toCharArray(buf,200);
  client.publish("pernec/celcom/pos/telemetry/POSBOX-004/",buf);
}

void detect()
{
val = digitalRead(sensor);   // read sensor value
  if (val == HIGH) {           // check if the sensor is HIGH
    digitalWrite(LED, HIGH);   // turn LED ON
    delay(100);                // delay 100 milliseconds

    if (state == LOW) {
      Serial.println("Motion detected!");
      Serial.println("==================================");
      execute();
      state = HIGH;       // update variable state to HIGH
    }
  }
  else {
      digitalWrite(LED, LOW); // turn LED OFF
      delay(200);             // delay 200 milliseconds

      if (state == HIGH){
        Serial.println("==================================");
        Serial.println("Motion stopped!");
        state = LOW;       // update variable state to LOW
    }
  }
}

void loop()
{
  detect();
  client.loop();
}

BR
Shamry