OLIMEX/NB-IoT #2: BC66 Unable to communicate with Linux over USB

spit4520, 2019-12-13:
Hello,

I am unable to connect with the device over usb and perform backup's or flash. I just bought the device and I got it today.

Please see the related ticket: https://github.com/Wiz-IO/Arduino-Quectel-BC66/issues/16

Thanks

DanKoloff, 2019-12-13:
Did you try holding PWR_KEY1 and/or RST_KEY to turn it on?

Also check here: https://github.com/OLIMEX/NB-IoT/blob/master/SOFTWARE/AT_example

spit4520, 2019-12-16:
Yes I have successfully flashed the board. I am trying to now restore the board to a state where I can write AT commands to it (I flashed the board before setting the APN by mistake) If you have any advice please advise 

see related issue: https://github.com/Wiz-IO/Arduino-Quectel-BC66/issues/16

spit4520, 2019-12-16:
Hello!

I have an update to the issue. I have been speaking with Georgi Angelov with his PlatformIO implementation of your board. Apparently on windows there is a second COM created that is for direct AT commands to the modem, but on Unix like devices with the CM34X driver I am only able to see /dev/ttyUSB0 and that is the serial/debugging interface NOT the AT command interface. He suggested I contact you directly for support on the Linux/MacOS implementation of the driver. I will road map the process for the software stack and add it into the Wiki so this issue is hopefully resolved in the future :) 

Let me know what the next steps would be!

Scott

DanKoloff, 2019-12-17:
Scott, we are writing a short guide on how to restore AT firmware. Should be done later today.

Yet, about the driver and the second serial interface, I believe it is not possible to get two interfaces via a driver update. CH340T has one serial interface, single-channel. You are probably comparing our design with boards that have dual-channel serial converters, which might have the option for two serial interfaces.

DanKoloff, 2019-12-17:
Try this: https://github.com/OLIMEX/NB-IoT/tree/master/SOFTWARE/AT_restore_tool

closed 2020-10-13 by DanKoloff