OLIMEX/NB-IoT #4: Getting error "Wrong SN number"

ASL07, 2020-02-27:
Hi, 

I have installed the drivers using the https://github.com/OLIMEX/ch340-dkms repo.
The install_ch340.sh script provided on this repo fails with the error `Expected ( at line 19`

The BC66 devkit is connected to the pc and detected as a serial port (ttyUSB0)
When I run REPAIR.sh I get the following error
`Wrong SN number`

kalata23, 2020-02-27:
Hi, ASL07,
Unfortunately there was issue with README not displaying the parrameters of 'REPAIR.sh' which I think is the case of your issue.
Syntax is: 

`./REPAIR.sh SN IMEI SerialPort`

Both **SN** and **IMEI** are written on the metal plate of the BC666 module. **Serial port** should be something like **/dev/ttyUSB0**, but the digit in the end can differ.

Command execution example:

`./REPAIR.sh MPA19BS1F020305 86787890123456123  /dev/ttyUSB0`

Please keep a note, that SN and IMEI numbers  you pass to the script file, must match with the ones written on the metal plate.

Kaloyan

ASL07, 2020-02-27:
Hi,
Do I have to press reset key or power key when I run the script?
I am pressing reset key after starting the script, but it fails most of the times. 
I was able to get `Removing app.......................[ DONE ]` only once, and I always get ` Restoring nvdm.....................[ FAILED ]
`

ASL07, 2020-02-27:
> Hi,
> Do I have to press reset key or power key when I run the script?
> I am pressing reset key after starting the script, but it fails most of the times.
> I was able to get `Removing app.......................[ DONE ]` only once, and I always get `Restoring nvdm.....................[ FAILED ]`

It works now. 
We need to press reset AFTER a couple of seconds after the script outputs `Removing app......................` or `Restoring nvdm.....................` . If the reset button is pressed too soon then it fails

kalata23, 2020-02-27:
Can you tell me the revision of the board that u have? You shouldn't be forced to press RST button.

ASL07, 2020-02-28:
Yes, if you tell me where to find it... The only thing I can see is 2019

kalata23, 2020-02-28:
> Yes, if you tell me where to find it... The only thing I can see is 2019

 It should be next to the BC66 module.

ASL07, 2020-02-28:
> > Yes, if you tell me where to find it... The only thing I can see is 2019
> 
> It should be next to the BC66 module.

It's Rev B

kalata23, 2020-02-28:
No need to manually reset the board with Rev B. The flash tool resets the board. Maybe you have bad USB connection (bad usb cable or not properly plugged).  Also to avoid reset issue, you can try to disconnect the board and connect it seconds before executing the REPAIR.sh script. Keep in mind that all Serial connections with the board should be closed before executing REPAIR.sh.

Kaloyan

ASL07, 2020-02-28:
> No need to manually reset the board with Rev B. The flash tool resets the board. Maybe you have bad USB connection (bad usb cable or not properly plugged). Also to avoid reset issue, you can try to [...]

I have tested this again
Disconnecting the board and reconnecting it just before the script is run helps, but it fails sometimes.
If the board is already connected, then I have to press the reset button for it to work

ASL07, 2020-08-24:
#3 
Hi,
Now the restore.sh script does not work at all
Previously, connecting the USB would make it work intermittently, but right now, it does not work at all.

Is there a way to restore AT functionality using IoT flash tool on Windows, or even Qflash?

I have a device backup but whenever I try to flash it with Qflash or flashtool I get "Format error"

kalata23, 2020-08-24:
> #3
> Hi,
> Now the restore.sh script does not work at all
> Previously, connecting the USB would make it work intermittently, but right now, it does not work at all.
>
> Is there a way to restore AT [...]

What version of the firmware did you use before uploading your own application?
Does restore.sh give you any error messages?

closed 2020-10-13 by DanKoloff