OLIMEX/NB-IoT #7: Cannot use 5V logic. Level shifter not working when jumper closed between 1-2 and open betwee 2-3

ASL07, 2020-06-16:
Hi,
I am using BC66 dev kit.
I need to interface a 5V logic device.
I have soldered the jumper between position 2 and 1 (VCCB), so now VCCB is supposed to be an INPUT, as per your documentation
I have connected both 5V_EXT and VCCB to 5V. 

Then, I am using a simple program which simply sets GPIO1 HIGH

Expected:
GPIO1 Voltage is 5V (VCCB)

Actual:
GPIO1 voltage is ~2.4V

If I leave VCCB unconnected (same jumper set-up), then GPIO1 is ~2.9V at high level

DanKoloff, 2020-06-17:
> I have soldered the jumper between position 2 and 1 (VCCB), so now VCCB is supposed to be an INPUT, as per your documentation

Did you cut the original connection between position 3 and 2?

ASL07, 2020-06-17:
Yes,
I have only connected position 1-2, position 2-3 are disconnected

closed 2026-09-24 by DanKoloff