OLIMEX/NB-IoT #8: GPIO HIGH level is 1.45V when VCCB is set to output (BC66 devkit)

ASL07, 2020-06-16:
Hello,

I have closed both position 1-2 and 2-3 of the VCCB/3v3 jumper.

Now, I have a simple test program which just sets GPIO1 to HIGH

I would expect GPIO  high level to be 3v3, however it is 1.45V

Could you tell me what the problem is, or help to troubleshooting it?

DanKoloff, 2020-06-17:
It is illegal to have both positions closed...

ASL07, 2020-06-17:
Hello, 

Your documentation says otherwise. It clearly says that in order to use the VCCB pin as 3v3 output, both positions of the jumper should be closed. Please see below:

https://github.com/OLIMEX/NB-IoT/blob/master/HARDWARE/NB-IoT-DevKit-pinout.png
﻿﻿
[image]

DanKoloff, 2020-06-17:
Yeah, this is true. I am testing it right now, my initial tests shows ~3.3V at GPIO1 with a board from the shop without any hardware or software changes. I power via the micro USB. After closing both jumpers I also measure 3.3V at VCCB pin. Will test with some toggle code now and will update you.

DanKoloff, 2020-06-17:
OK, so it seems I can replicate the problem for both this issue and https://github.com/OLIMEX/NB-IoT/issues/7

It doesn't occur if I use oscilloscope to measure but it appears if I use multimeter. In fact the issue appears if I use three different multimeter tools and never occurs with the oscilloscopes, it also doesn't happen if I use 10 MOhm resistor. I shot a video with the oscilloscope:

https://www.youtube.com/watch?v=6IA7m9HBqNU

I believe the TXB0108PWR gets somehow disturbed by the multimeter tool, but can't find anything in the datasheet so far.

ASL07, 2020-06-17:
Hello,
Thanks for your testing
Now that you mention this, I was experiencing very weird behaviour when measuring GPIO1 with a portable scope. The Voltage level on GPIO1 would change whenever I disconnected and reconnected the scope...

I thought it could be a bad reading from the scope, but now that you mention the multimeter causes disturbances, it could also be that connecting the scope causes the voltage to change...

Will do more tests later on and update this issue

Please let me know if you make any more progress

DanKoloff, 2020-06-18:
> Btw, have you been able to get the GPIO working at 5V? By using VCCB as input and providing 5V_EXT, as I mentioned in #7 ?

Yes. Same behavior exactly. No multimeter attached = no problem. As soon as I attach the multimeter voltage goes down. It is the same if I kept it during power up.

No issues if I use oscilloscope tho or if I attach 10 mega ohm resistors instead.

ASL07, 2020-06-18:
I have connected a 5V logic sensor (HC-SR04) to the module and I can interface it properly.

It is really weird that it only happens when you connect a multimeter

DanKoloff, 2020-06-18:
I also don't have idea why it happens, but I am glad we discovered it.

DanKoloff, 2020-06-19:
We investigated further. It seems the multimeters have high capacitance and the TXB0108PWR buffer bugs at around 70pF as per datasheet:

"The TXB0108 is designed to drive capacitive loads of up to 70 pF. The output drivers of the TXB0108 have low dc drive strength. If pullup or pulldown resistors are connected externally to the data I/Os, their values must be kept higher than 50 kΩ to ensure that they do not contend with the output drivers of the TXB0108."

You can now find the documents here: https://github.com/OLIMEX/NB-IoT/tree/master/DOCs

There is a workaround if you want to measure with such tool - add a resistor in series between the board and the tool. I used 200 kOhm resistor and this allows the measurement.

closed 2026-09-24 by DanKoloff