OLIMEX/NB-IoT #9: Board is not powering up unless USB is connected

ASL07, 2020-08-24:
Hello,

I am having a strange issue with my board.

I followed this and have connected position 1-2 and disconnected position 2-3 of the jumper, in order to set VCCB as power input
[image]

I expect the board to power on whenever I connect 5V to VCCB, but it doesn't power up unless I connect the USB as well

DanKoloff, 2020-08-24:
Did you provide 5V to pin #1 +5V_EXT too? The jumpers and VCCB power input set the buffers but you still need to provide input +5V_EXT.

Important! When you provide 5V to VCCB and 5V to the input all GPIOs would become 5V. Make sure you don't have 3.3V peripherals connected to GPIOs or they might get damaged.

ASL07, 2020-08-24:
Hi,
I should have mentioned. 
Yes am providing 5V to both 5V_EXT and VCCB
And yes, I want all GPIO to have 5V logic

ASL07, 2020-08-24:
Sorry, my fault, it was a power supply issue. I switched to a more reliable one and now it works.

closed 2020-08-24 by ASL07