OLIMEX/NB-IoT #11: AT Commands from external MCU

Simonefardella, 2020-10-02:
Hi, it's possible to interface this module with external MCU and send AT Commands from (for example) an Arduino or ESP32?

It's possible to have some examples?
Thanks

DanKoloff, 2020-10-13:
Yes, you can control the board from another board via AT commands. For initial evaluation I highly recommend starting with NB-IoT-DevKit or NB-IoT-DevKit-ANT before switching to the designs more suitable for embedding:

https://www.olimex.com/Products/IoT/NB-IoT/NB-IoT-DevKit/open-source-hardware

Simonefardella, 2020-10-27:
Thanks, about the serial AT commands, i have to wire my uart to the PIN 17-18 (TxD-RxD GPIO)?
Thanks

DanKoloff, 2020-10-28:
17 and 18. In NB-IoT-DevKi we have placed the CH340T USB-UART converter to pins 17 and 18.

Simonefardella, 2020-10-28:
Thanks man, i'll close the issue.

closed 2020-10-28 by Simonefardella

artursmet, 2020-11-08:
@DanKoloff Is it possible to use NB-IoT-DevKit (with that USB-UART converter) in the scenario mentioned by @Simonefardella ?
I'd like to do exactly the same thing and I already have the NB-IoT-Devkit. Is there a way to achieve the AT commands control from external MCU using that board?

DanKoloff, 2020-11-09:
Yes, it should be possible but I am not sure of the specifics since I haven't tested it. Few things to consider - USB-UART converter should be at 3.3V DC level; maybe if using the 17/18 pins it would cause problems with the CH340T converter that is placed on the same lines, if that is the case you might want to keep the USB disconnected (or the USB data lines if that is used for powering), if that is not working maybe you would need to isolate CH340T from interfering, but maybe there is a way to set second UART (28/29) for the AT so you don't interfere with the CH340T but not sure how it can be done can't find AT command for it, maybe ask Quectel if there is such firmware (set for pins 28/29).

artursmet, 2020-11-09:
@DanKoloff Thanks for the answer - I'll try contacting Quectel about using the second UART for AT commands.
It looks like the CH340T is interfering still - I'm powering the devkit over `+5V_Ext` pin, the USB is not connected at all.
I've seen the #9 though - should I switch the jumper and provide current also to VCCB, in order to power up the board from external (non USB) power source?

DanKoloff, 2020-11-09:
If you are doing that it should work. Few missteps could have happened: USB-serial RX and TX not being at 3.3V; not have GND of power supply and GND of USB-serial cable connected to board; forgetting to power the board after attaching power supply with the power button; not sending proper string from host board (I always forget about CR LF around AT boards, not sure which or if any of proper ending is required).

artursmet, 2020-11-10:
@DanKoloff Thanks for your help I really appreciate it.
Despite the good wiring, I was unable to use the main ports in my setup. However I had contacted Quectel support and they have shared with me a very helpful hints. 

The BC66 supports AT commands only on one UART interface at a time. There is a way (internal AT command) that can be used to switch the AT command interface.
Sharing the syntax:

```
AT+EPORT=1,connl,1
```
first attribute means that we want to switch the interface
`connl` means AT command interface
third attribute is the desired interface which we'd like to use. 
```
      0 : UART0 (MAIN)
      1:  UART1 (AUX)
      2:  UART2
      3:  UART3
      4:  USB
      5:  USB
```

Also if you want to switch the hardcoded baud rate, it's possible via

```
AT+EPORT=3,1,9 
```
3 - the command will be used in changing baud rate mode
1 - the port from above map (UART AUX in my example)
9 - desired baud rate value (map below)

[image]

I really recommend contacting Quectel support, they're very responsive.

In my final setup I power up the BC66 devkit on the `+5V_Ext` port, I connect the serial interface on pins 28/29 and I can control the module with AT commands.

DanKoloff, 2020-11-11:
Thank you very much for the update, pretty sure other people would find it helpful in future too! Glad the issue had been solved in such an elegant way! 

My experience with Quectel support had always been similarly positive.