OLIMEX/NB-IoT #13: Unable to compile example projects (missing os_wizio.h)

belveder79, 2021-01-13:
I followed the install instructions for the BC66 module (the one without USB connector) but none of the examples compile as I end up with:

`/Users/arth/.platformio/packages/framework-quectel/arduino/cores/bc66/interface.h:27:10: fatal error: os_wizio.h: No such file or directory`

I'm using latest Atom and PlatformIO on OSX.

DanKoloff, 2021-01-14:
It can be something related to the OSX installation. I think Wiz-IO didn't fully test on a OSX, based on his comment in the description ("test and report "):

https://github.com/Wiz-IO/platform-quectel

belveder79, 2021-01-14:
it is also broken on Windows

DanKoloff, 2021-01-14:
I've update the guide to list proper addresses now. Not sure if the examples re-hosted at our GitHub would work though, maybe use the default ones.

belveder79, 2021-01-14:
Ok, so I got the samples to compile, but I can't get the device to connect at all (just crambled chars - see Olimex Forum post) Anyway, probably there is some issue with the baud rate on the preinstalled image, so I could probably fix it by setting Serial baud rate manually in the upload image. However, it is a bit frustrating as there is essentially no dedicated documentation for the module at all - I have no idea where to start... 

Let's say the Arduino examples from Wiz-IO are compiling - how do I get them onto the board? I need to get the board into some bootloader state by pressing buttons, but how to do that here?

DanKoloff, 2021-01-15:
You need DTR and DTS, inspect the schematic of NB-IoT-DevKit:

https://github.com/OLIMEX/NB-IoT/blob/master/HARDWARE/NB-IoT-DevKit_Rev_B/NB-IoT-DevKit_Rev_B.pdf

You can use ESP-PROG for the task:

https://www.olimex.com/Products/IoT/Programmer/ESP-PROG/open-source-hardware

 or use the schematic as basis to understand better:

https://www.olimex.com/Products/IoT/Programmer/ESP-PROG/open-source-hardware

belveder79, 2021-01-15:
ok, the point is I'm not using the DevKit, I'm using the breakout board:

https://www.olimex.com/Products/IoT/NB-IoT/NB-IoT-BC66/resources/NB-IoT-BC66-pinout.pdf
https://www.olimex.com/Products/IoT/NB-IoT/NB-IoT-BC66/resources/NB-IoT-BC66_Rev_B.pdf

So as far as I get the schematics, 
- 3.3V is actually fine as power
- level shifters in place should translate 1.8V to 3.3V already but NOT for RxD_main and TXD_main (?)
- I should connect the UART to RxD_main and TxD_main rather than RxD_dbg and TXD_dbg, but need to have my own level shifters in place (?)

DanKoloff, 2021-01-20:
Judging by this forum message the issue is now solved:

https://www.olimex.com/forum/index.php?topic=8037.msg30338#msg30338

However, I am not sure 100% what exactly was the fix - was it solved by using the TXD_MAIN and RXD_MAIN pins (instead of the dbg ones)?

belveder79, 2021-01-24:
yes, actually it was resolved by using the main RX/TX pins. I did not really test flashing the module yes, but merely playing with AT commands...

DanKoloff, 2021-01-27:
Alright thank you for the feedback, I am sure your feedback would help other people in future. Closing the issue now as resolved.

closed 2021-01-27 by DanKoloff